laurencelundblade/QCBOR #69: 3 Tests Failing on ARM Cortex-M4 CPU

dbeckwith01, 2020-12-08:
I'm trying to run QCBOR on an ARM Cortex-M4 CPU and the following 3 failures.

`NotWellFormedTests FAILED (returned 2079)ParseMapAsArrayTest PASSED`
`DecodeFailureTests FAILED (returned 9620)
`BstrWrapErrorTest FAILED (returned -3)`

I'm using the latest master branch c02e13eb6263381195290424c4cdffd440a0aa96 and I have the following configurations defined
`QCBOR_DISABLE_FLOAT_HW_USE`
`USEFULBUF_CONFIG_LITTLE_ENDIAN`
`QCBOR_DISABLE_INDEFINITE_LENGTH_STRINGS`
`QCBOR_DISABLE_PREFERRED_FLOAT`

I've tried to determine the cause of the failures but I haven't been able to pin it down.

Thanks

laurencelundblade, 2020-12-08:
Hey Darren,

I'll look at this, but it may take a few days. This certainly should be tracked down and fixed. Thanks for reporting.

This is 32-bit, right?

dbeckwith01, 2020-12-08:
Hi Laurence,

Thanks for getting back to me.  Yes, this is a 32-bit processor with limited FLASH and RAM, 1MB and 256KB.  I say a post by Jared Wolff about using it.

Any thoughts on how QCBOR compares to TinyCBOR?

laurencelundblade, 2020-12-09:
Next questions:
what compiler?
what is SIZE_MAX?

dbeckwith01, 2020-12-09:
The compiler is from part of Segger Embedded Studio https://www.segger.com/products/development-tools/embedded-studio/

SIZE_MAX is set to INT32_MAX or 2147483647L

laurencelundblade, 2020-12-10:
I just pushed a fix that should take care of these two:
  NotWellFormedTests FAILED (returned 2079)
  DecodeFailureTests FAILED (returned 9620)

Nothing wrong with the actual implementation, just an issue with the way the tests were constructed. They expected SIZE_MAX to be at least UINT32_MAX.

I don't know about this yet:
  BstrWrapErrorTest FAILED (returned -3)`
The version I just pushed will report more detail on that failure that might help me figure it out. It doesn't seem to be related to SIZE_MAX. Would appreciate if you re run the tests and tell me the results.

laurencelundblade, 2020-12-10:
I haven't looked very hard a tinyCBOR and when I did it was a while a go. Here's a few things I've tried to do in QCBOR:
- Clear documented public interface. You don't have to look at the internal code to use QCBOR
- Handle CBOR tag correctly. A messy topic that was only recently clarified. See RFC 8949.
- Easy install without #ifdefs 
- Provide common code for common things needed when decoding, particularly map decoding

laurencelundblade, 2020-12-12:
I've pushed another fix that may solve the problem with  BstrWrapErrorTest FAILED (returned -3)

I think it's related to SIZE_MAX < UINT32_MAX.

dbeckwith01, 2020-12-14:
@laurencelundblade Thanks for adding the updates.  I should have time to test this today.

laurencelundblade, 2021-02-11:
Fixes were made that should have addressed the issue, so closing after waiting a month for confirmation.

closed 2021-02-11 by laurencelundblade