laurencelundblade/QCBOR #75: const-bursting code creates warnings that really are an issue

mcr, 2021-01-26:
The way that UsefulBufC is used as a result is a real problem.
It takes  const item, and then changes it.  This violates the const guarantees that optimizers use.  
I see no advantage to this part.  I propose to remove UsefulBufC, and just #define UsefulBufC UsefulBuf.

laurencelundblade, 2021-02-21:
The const-ness of UsefulBufC is a core part of the intentional design. We have const char * so there should be const byte strings. I wouldn't want to give that up without a good reason.

I would think having more things const helps the compiler optimize.

From your other PR, I suspect this relates to -Wcast-qual. Is that right? Is there any particular reason this warning is important other than basic general code hygiene? I want to be sure I understand what's going on here.

mcr, 2021-02-21:
Yes, const helps the compiler optimize, as long as the code follows the contract, and in the case of using a UsefulBufC as a result parameter, that definitely breaks things.

For instance in the case of: 

const int A = 10;
function(&A);
B=A+1;

The compile is free to optimize "B=A+1" to be "B=11", because A is a constant, and can not change.
If function uses a const busting pragma/cast to change A, which means that the optimized code is incorrect.
This is not about warnings.  This is about correctness.
I have been down this rabbit hole 20 years ago with C, and with multiple PhD compiler designers. We tried to make certain code make sense, we ought to be able to mark certain things as const, but often you just can't.

laurencelundblade, 2021-02-21:
Sorry, if I'm being dense, but I don't know what is broken yet. How is this broken

`QCBOREncode_Finish(QCBOREncodeContext *pCtx, UsefulBufC *pEncodedCBOR);
`

but this

`QCBOREncode_Finish(QCBOREncodeContext *pCtx, const char ** pEncodedCBOR, size_t * pnEncodedCBORsize);
`

is not broken? Or are they both broken?

I'm not sure what the breakage / incorrectness is. 

Is it just that a compiler can't optimize this? I'd be a bit surprised at that. LLVM and GCC are very good these days, better than old compilers. I've looked at their output a little for usage of UsefulBuf and UsefulBufC as a parameter and return value and it seems OK.

mcr, 2021-02-21:
Laurence Lundblade <notifications@github.com> wrote:
    > Sorry, if I'm being dense, but I don't know what is broken yet. How is
    > this broken

    > `QCBOREncode_Finish(QCBOREncodeContext *pCtx, UsefulBufC
    > *pEncodedCBOR); `

    > but this

    > `QCBOREncode_Finish(QCBOREncodeContext *pCtx, const char **
    > pEncodedCBOR, size_t * pnEncodedCBORsize); `

    > is not broken? Or are they both broken?

Both are broken.
The const char is promising not to modify pEncodedCBOR contents.
But, it does.

    > I'm not sure what the breakage / incorrectness is.

    > Is it just that a compiler can't optimize this? I'd be a bit surprised
    > at that. LLVM and GCC are very good these days, better than old
    > compilers. I've looked at their output a little for usage of UsefulBuf
    > and UsefulBufC as a parameter and return value and it seems OK.

No, it's that the compiler that optimizes this does the wrong thing.
Since the bytes can't change, the compiler is free to move .._Finish after
the code that examines the content.

--
]               Never tell me the odds!                 | ipv6 mesh networks [
]   Michael Richardson, Sandelman Software Works        |    IoT architect   [
]     mcr@sandelman.ca  http://www.sandelman.ca/        |   ruby on rails    [

laurencelundblade, 2021-02-22:
I can't imagine there's danger of compilers generating incorrect code, code that doesn't execute correctly, because of the const. Thousands of C programmers would have noticed.

So maybe it is not as const as expected. It is still more const that it would be without the const declaration so it is still doing good for the coder.

Can you give a concrete example of some code where something breaks? Then tell me why removing the const would result in less breakage.

mcr, 2021-02-22:
Hi, I guess I will fork QCBOR then.
I don't have time to read the C-specification to you.

closed 2021-02-22 by mcr