laurencelundblade/QCBOR #78: possible problems with empty map in map  --- how to submit unit test

mcr, 2021-02-18:
I am trying to decode the following map:

````
A1    # map(1)
   02 # unsigned(2)
   A0 # map(0)
````

i.e.: ````{2: {}}````

It seems that QCBORDecode_GetNext(&qdc, &item) returns with item.label.int64 as zero.
I think that this might be because of the empty map as the value.  I'm not really sure.

I guess an example ought to go into tests/qcbor_decode_tests.c, but it would sure be nicer if this could be a single test per file somehow.

laurencelundblade, 2021-02-18:
Hi Michael,

I'll look at this in the next day or two. It is likely a bug as I don't recall testing this and there are a lot of corner cases related to empty maps and arrays.

I don't have any file I/O in all of QCBOR for a reason. Some embedded systems don't have files. I'll debug this by making a test case and setting a break point at it (in Xcode).

Thanks for reporting.

mcr, 2021-02-18:
There is no makefile in the tests directory, and no "make check" in the main makefile.
It would be great if it had that, then we could run the whole thing under travis-ci.

mcr, 2021-02-18:
I changed my code to just put a 1 into the target, rather than an empty map: 0xa1	0x01	0x01
SAME THING.

mcr, 2021-02-18:
#81 shows this problem against the top of your tree.
I'm perplexed.  I am decoding multiple items without a problem.
I have also tried adding a second entry to the map.

mcr, 2021-02-18:
Well, I screwed up my unit test case.  Standby... :-)

mcr, 2021-02-18:
okay, fixed: I compared to TYPE_ARRAY, and that was wrong.
But, a printf():

````
   printf("type: %d nest: %d labeltype: %d label: %d\n",
          Item.uDataType,
          Item.uNestingLevel,
          Item.uLabelType,
          Item.label.int64);
```

shows me:
type: 5 nest: 0 labeltype: 0 label: 0

the labeltype being 0 is weird.

mcr, 2021-02-18:
My test case was inaccurate: I simplified too much.  But my underlying issue remains in my code.

mcr, 2021-02-19:
Further: -DCONFIG_ variables not set consistently across users of QCBOR includes.

closed 2021-02-19 by mcr