laurencelundblade/QCBOR #129: Uninitialized data warnings

thewtex, 2022-02-17:
With GCC 10.2.1 / Linux:

```
/home/matt/src/QCBOR/src/qcbor_decode.c: In function ‘QCBORDecode_GetTaggedStringInMapSZ’:
/home/matt/src/QCBOR/src/qcbor_decode.c:3180:9: warning: ‘Item.uDataType’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 3180 |       if(uDataType == puTypeList[i]) {
      |         ^
/home/matt/src/QCBOR/src/qcbor_decode.c:3287:14: note: ‘Item.uDataType’ was declared here
 3287 |    QCBORItem Item;
      |              ^~~~
/home/matt/src/QCBOR/src/qcbor_decode.c:3201:80: warning: ‘Item.uTags[0]’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 3201 |    if(!(TagSpec.uTagRequirement & QCBOR_TAG_REQUIREMENT_ALLOW_ADDITIONAL_TAGS) &&
/home/matt/src/QCBOR/src/qcbor_decode.c:3287:14: note: ‘Item.uTags[0]’ was declared here
 3287 |    QCBORItem Item;
      |              ^~~~
/home/matt/src/QCBOR/src/qcbor_decode.c: In function ‘QCBORDecode_EnterBstrWrappedFromMapN’:
/home/matt/src/QCBOR/src/qcbor_decode.c:3201:80: warning: ‘Item.uTags[0]’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 3201 |    if(!(TagSpec.uTagRequirement & QCBOR_TAG_REQUIREMENT_ALLOW_ADDITIONAL_TAGS) &&
/home/matt/src/QCBOR/src/qcbor_decode.c:3695:14: note: ‘Item.uTags[0]’ was declared here
 3695 |    QCBORItem Item;
      |              ^~~~
/home/matt/src/QCBOR/src/qcbor_decode.c:3209:14: warning: ‘Item.uDataType’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 3209 |    const int nItemType = pItem->uDataType;
      |              ^~~~~~~~~
/home/matt/src/QCBOR/src/qcbor_decode.c:3695:14: note: ‘Item.uDataType’ was declared here
 3695 |    QCBORItem Item;
      |              ^~~~
/home/matt/src/QCBOR/src/qcbor_decode.c: In function ‘QCBORDecode_EnterBstrWrappedFromMapSZ’:
/home/matt/src/QCBOR/src/qcbor_decode.c:3180:9: warning: ‘Item.uDataType’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 3180 |       if(uDataType == puTypeList[i]) {
      |         ^
/home/matt/src/QCBOR/src/qcbor_decode.c:3713:14: note: ‘Item.uDataType’ was declared here
 3713 |    QCBORItem Item;
      |              ^~~~
/home/matt/src/QCBOR/src/qcbor_decode.c:3201:80: warning: ‘Item.uTags[0]’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 3201 |    if(!(TagSpec.uTagRequirement & QCBOR_TAG_REQUIREMENT_ALLOW_ADDITIONAL_TAGS) &&
/home/matt/src/QCBOR/src/qcbor_decode.c:3713:14: note: ‘Item.uTags[0]’ was declared here
 3713 |    QCBORItem Item;
      |              ^~~~
/home/matt/src/QCBOR/src/qcbor_decode.c: In function ‘QCBORDecode_GetBoolInMapSZ’:
/home/matt/src/QCBOR/src/qcbor_decode.c:2747:4: warning: ‘*((void *)&Item+48)’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 2747 |    memcpy(pMe->uLastTags, pItem->uTags, sizeof(pItem->uTags));
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/matt/src/QCBOR/src/qcbor_decode.c:3815:14: note: ‘*((void *)&Item+48)’ was declared here
 3815 |    QCBORItem Item;
      |              ^~~~
/home/matt/src/QCBOR/src/qcbor_decode.c:3763:4: warning: ‘Item.uDataType’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 3763 |    switch(pItem->uDataType) {
      |    ^~~~~~
/home/matt/src/QCBOR/src/qcbor_decode.c:3815:14: note: ‘Item.uDataType’ was declared here
 3815 |    QCBORItem Item;
      |              ^~~~
/home/matt/src/QCBOR/src/qcbor_decode.c: In function ‘QCBORDecode_GetEpochDateInMapSZ’:
/home/matt/src/QCBOR/src/qcbor_decode.c:1947:44: warning: ‘Item.uDataType’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 1946 |          const double d = pDecodedItem->uDataType == QCBOR_TYPE_DOUBLE ?
      |                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1947 |                    pDecodedItem->val.dfnum :
      |                    ~~~~~~~~~~~~~~~~~~~~~~~~^
 1948 |                    (double)pDecodedItem->val.fnum;
      |                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/matt/src/QCBOR/src/qcbor_decode.c:3900:14: note: ‘Item.uDataType’ was declared here
 3900 |    QCBORItem Item;
      |              ^~~~
/home/matt/src/QCBOR/src/qcbor_decode.c: In function ‘QCBORDecode_GetEpochDaysInMapSZ’:
/home/matt/src/QCBOR/src/qcbor_decode.c:3180:9: warning: ‘Item.uDataType’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 3180 |       if(uDataType == puTypeList[i]) {
      |         ^
/home/matt/src/QCBOR/src/qcbor_decode.c:3998:14: note: ‘Item.uDataType’ was declared here
 3998 |    QCBORItem Item;
      |              ^~~~
/home/matt/src/QCBOR/src/qcbor_decode.c:3948:12: warning: ‘Item.val.epochDays’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 3948 |    *pnDays = pItem->val.epochDays;
      |    ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/home/matt/src/QCBOR/src/qcbor_decode.c:3998:14: note: ‘Item.val.epochDays’ was declared here
 3998 |    QCBORItem Item;
      |              ^~~~
/home/matt/src/QCBOR/src/qcbor_decode.c: In function ‘QCBORDecode_GetBignumInMapSZ’:
/home/matt/src/QCBOR/src/qcbor_decode.c:3180:9: warning: ‘Item.uDataType’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 3180 |       if(uDataType == puTypeList[i]) {
      |         ^
/home/matt/src/QCBOR/src/qcbor_decode.c:4117:14: note: ‘Item.uDataType’ was declared here
 4117 |    QCBORItem Item;
      |              ^~~~
/home/matt/src/QCBOR/src/qcbor_decode.c:3201:80: warning: ‘Item.uTags[0]’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 3201 |    if(!(TagSpec.uTagRequirement & QCBOR_TAG_REQUIREMENT_ALLOW_ADDITIONAL_TAGS) &&
/home/matt/src/QCBOR/src/qcbor_decode.c:4117:14: note: ‘Item.uTags[0]’ was declared here
 4117 |    QCBORItem Item;
      |              ^~~~
/home/matt/src/QCBOR/src/qcbor_decode.c: In function ‘QCBORDecode_GetDoubleConvertAll’:
/home/matt/src/QCBOR/src/qcbor_decode.c:5326:14: warning: ‘Item.uDataType’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 5326 |    QCBORItem Item;
      |              ^~~~
/home/matt/src/QCBOR/src/qcbor_decode.c:5268:36: warning: ‘Item.val.expAndMantissa.nExponent’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 5268 |             *pdValue = dMantissa * pow(10, (double)pItem->val.expAndMantissa.nExponent);
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/matt/src/QCBOR/src/qcbor_decode.c:5326:14: note: ‘Item.val.expAndMantissa.nExponent’ was declared here
 5326 |    QCBORItem Item;
      |              ^~~~
/home/matt/src/QCBOR/src/qcbor_decode.c:5326:14: warning: ‘Item.val.bigNum.len’ may be used uninitialized in this function [-Wmaybe-uninitialized]
/home/matt/src/QCBOR/src/qcbor_decode.c:5326:14: warning: ‘Item.val.expAndMantissa.Mantissa.bigNum.len’ may be used uninitialized in this function [-Wmaybe-uninitialized]
```

laurencelundblade, 2022-02-20:
Looks like gcc 10.2 has some warnings that 10.1 doesn't.

I don't think there are any real problems in QCBOR, but I will figure out a fix in the next few days.

Thanks for reporting!

laurencelundblade, 2022-03-02:
I've spent some time on this and so far can't replicate it and haven't got to the bottom of it. 

When I try adding -Wmaybe-uninitialized for gcc-9, gcc-10 and gcc-11 on MacOS it tells me the warning flag doesn't exist. I've also tried llvm/clang.

I build with -Wall and -Wextra and that is supposed tot set -Wmaybe-uninitialized, but I don't get warnings.

I do not believe there are any actual problems with the code.

I will keep looking, but if you have any more info or such it might be helpful.

laurencelundblade, 2022-08-14:
Closing this because I can't reproduce it, no one else has turned this and the filer of this hasn't commented further.

closed 2022-08-14 by laurencelundblade