laurencelundblade/QCBOR #139: BstrWrapErrorTest failing on ARM FVP AN521 (release v1.0)

adam2809, 2022-05-12:
I am working on the TF-M project and trying to switch from a version of qcbor which was copied to the repo to an upstream version (v1.0 release). The test fails because the error is QCBOR_ERR_BUFFER_TOO_SMALL instead of QCBOR_ERR_ARRAY_NESTING_TOO_DEEP. The latest working version I managed to find is 8c858ab81c9c4bd85d42fff3c465e4abfc59721d the next commit fails. I found that on the second last iteration of test/qcbor_encode_tests.c:1820 the EC.nesting.pCurrentNesting->uStart goes to 2831 which causes the if at src/UsefulBuf.c:248 to go in. You can try reproducing the issue using https://www.qemu.org/docs/master/system/arm/mps2.html mps2-an521. Let me know if you have issues reproducing this problem and I will help you.

laurencelundblade, 2022-05-14:
Yup, looks like something is wrong with the error codes in this area. Looking at it. Thanks.

Which compiler and CPU architecture?  I work on MacOS, so I can't run QEMU.

adam2809, 2022-05-16:
Thanks a lot for the quick response I checked and confirm that everything passes on guard-fix branch. I was using Arm Clang and Arm architecture.