laurencelundblade/QCBOR #143: Build error if ARM microlib is used

escherstair, 2022-05-16:
I know ARM microlib is not compliant with the ISO C library standard, and so I don't know how much interest there is in having QCBOR working if ARM microlib is used.
qcbor_decode.c includes `<fenv.h>` to use `feclearexcept()` and `fetestexcept()`
https://github.com/laurencelundblade/QCBOR/blob/9123e5b11f859249f97770572e3e02d23036ab08/src/qcbor_decode.c#L44
The problem  is that microlib does not support `feclearexcept()` and `fetestexcept()` (see [here](https://www.keil.com/support/man/docs/armlib/armlib_chr1358938938431.htm)).
Is it possible to add an
`#ifdef __MICROLIB`
to behave in different way if microlib is used?

laurencelundblade, 2022-05-17:
Have you tried defining QCBOR_DISABLE_FLOAT_HW_USE? That situation is its main purpose. You'll lose a little floating-point functionality.

Definitely want QCBOR to work with lots and lots of OS's and environments!

escherstair, 2022-05-17:
I thought to that, but it sounds strange to me because my hardware platfrom has full hardware support for floating point. It's only microlib that has `<fenv.h>`, but lacks of `feclearexcept()` and `fetestexcept()`.
Reading carefully [here](https://github.com/laurencelundblade/QCBOR#define-qcbor_disable_float_hw_use) I agree that this is the best solition.
And it works.

laurencelundblade, 2022-05-17:
Thanks for reporting back!

closed 2022-05-17 by laurencelundblade