laurencelundblade/QCBOR #152: UsefulBuf.h gives warnings when used with MSVC

dthaler, 2022-07-23:
UsefulBuf.h(677,9): warning C4068: unknown pragma 'GCC'
UsefulBuf.h(678,9): warning C4068: unknown pragma 'GCC'
UsefulBuf.h(680,9): warning C4068: unknown pragma 'GCC'
UsefulBuf.h(1760,9): warning C4068: unknown pragma 'GCC'
UsefulBuf.h(1761,9): warning C4068: unknown pragma 'GCC'
UsefulBuf.h(1763,9): warning C4068: unknown pragma 'GCC'

Any GCC pragmas should be inside something like
#ifdef __GNUC__
...
#endif

laurencelundblade, 2022-07-23:
Hi Dave -- it was left like this because llvm/clang and all the variants from ARM, Nvidia and such based on it also support these pragmas. Will see what I can do...

laurencelundblade, 2022-08-06:
Thanks for the PR!

laurencelundblade, 2022-08-06:
Why can't I close this?

laurencelundblade, 2022-08-06:
Can I close using Firefox?

closed 2022-08-06 by laurencelundblade