laurencelundblade/QCBOR #175: How to use cmake's disable float options?

toge, 2023-01-21:
In the top-level CMakeLists.txt, it appears to be removing the math lib link only when QCBOR_OPT_DISABLE_FLOAT_HW_USE is True.

```cmake
    # Using GCC
    target_link_libraries(qcbor
        PRIVATE
            $<$<NOT:$<BOOL:${QCBOR_OPT_DISABLE_FLOAT_HW_USE}>>:m>
    )
```

Assuming this implementation is ok, would you please tell me the correct usage of disable float in CMake?
From reading the README.md, I didn't clearly understand how to use the three options.

- If I set QCBOR_OPT_DISABLE_FLOAT_ALL to True, do I also need to specify QCBOR_OPT_DISABLE_FLOAT_HW_USE and QCBOR_OPT_DISABLE_FLOAT_PREFERRED to True?
- If QCBOR_OPT_DISABLE_FLOAT_PREFERRED is set to True, does QCBOR_OPT_DISABLE_HW_USE also need to be set to True?

toge, 2023-01-23:
@laurencelundblade 
Thank you for your response.
I have confirmed that the three options in CMake change the behavior and code size. It is a great code.

I wanted to achive two goals(those are independent)
- Eliminate library dependencies
- Minimize code size

To achieve these, would it be adequate to specify just one option each as follows?
- Eliminate library dependencies -> QCBOR_DISABLE_FLOAT_HW_USE
- Minimize code size -> QCBOR_DISABLE_PREFERRED_FLOAT

laurencelundblade, 2023-01-23:
Yes, those two will achieve what you want. You will still be able to encode and decode floating point numbers in a very basic way too.

You can also try USEFULBUF_DISABLE_ALL_FLOAT and see if that makes code even smaller. It probably doesn't, but you could try it. With the defined you cannot do anything with floating point numbers in QCBOR at all.

toge, 2023-01-24:
@laurencelundblade 
Now I understand.
Thanks a lot!

closed 2023-01-24 by toge