laurencelundblade/QCBOR #180: Backward compatibility is broken

tomoveu, 2023-02-15:
[This commit](https://github.com/laurencelundblade/QCBOR/commit/4f95c418e5abbe1567d3685c009afc5f2525edac#diff-7723a0dc9dbb26edf4d6397337bd2a9ceaf3530cd8b30d8fee708a6c292356e3) introduced an easy and simple way to have backward compatibility. 

Sometime after, this was removed without any explanation. 

Please restore backward compatibility:

```
/**
  * @file qcbor.h
  *
  * Backwards compatibility for includers of qcbor.h (which has been split
  * into four include files).
  */

 #include "qcbor/qcbor_encode.h"
 #include "qcbor/qcbor_decode.h"
```

Let me know if I can help in any way.

/Dimi

laurencelundblade, 2023-02-15:
See #181

tomoveu, 2023-02-15:
@laurencelundblade thank you for the quick response :+1:  This would be great. 

I will make sure to close this issue when PR 181 lands in master.

laurencelundblade, 2023-02-22:
Fixed by #181

closed 2023-02-22 by laurencelundblade

tomoveu, 2023-02-22:
Thank you @laurencelundblade 💯