laurencelundblade/QCBOR #329: QCBOREncode_FinishGetSize not usable if the original buffer is too small

BrianSipos, 2025-08-07:
I would like to use a simple pattern to encode variable-length PDUs with an optimistic initial size estimate to avoid double-encoding under good conditions.

The pattern is:
 1. Allocate a small fixed-size buffer
 2. Encode to that buffer as normal
 3. Use `QCBOREncode_FinishGetSize()` to determine if the buffer was large enough or not
 4. If it was not, allocate a properly sized (larger) buffer
 5. Encode to the dynamically sized buffer
 6. Use `QCBOREncode_Finish()` to finalize

My problem is currently that `QCBOREncode_FinishGetSize()` only sets the encoded size if the error state is `QCBOR_SUCCESS` but not if it is `QCBOR_ERR_BUFFER_TOO_SMALL`.

https://github.com/laurencelundblade/QCBOR/blob/9f09f0ca523d0a6d7b7870a6a3c1adb972ebc546/src/qcbor_encode.c#L1081-L1083

Is there a reason for that condition to be present in FinishGetSize at all?
Let the caller check the error code to see if the size is meaningful in its context.

laurencelundblade, 2025-08-09:
Hi Brian, I'll look at it some more, but encoding stops immediately when an error occurs so it doesn't calculate the whole length. There are also errors that are not due to the length. Maybe there can be an exception for QCBOR_ERR_BUFFER_TOO_SMALL, but there will have to be a lot of checking and testing to be sure it's OK and it may increase code size.

BrianSipos, 2025-08-09:
Thank you. Yes, this would be to treat QCBOR_ERR_BUFFER_TOO_SMALL the same condition as using the zero-length buffer to obtain the full size (which currently exists).

laurencelundblade, 2025-08-14:
Still working on this. It's a significant change to UsefulBuf error semantics so I'm hesitant.